shining-man/bsc_fw #235: Zusatzparameter oder Option bei Funktion 'Charge Current Cut Off'

mmoe-milano, 2025-03-15:
Mit Hilfe der Erweiterung sollte es möglich sein den 'setpoint-cv' Parameter schrittweite über die Zeit 
zu steuern. Zielsetzung: Vermeidung sehr große Ausgleichströme bei Victronanlagen unter Verwendung der Cut Off 
Funktion bei gleichzeitig aktiviertem DC-FeedIn.

Fuendiger, 2025-03-15:
Was genau bedeutet Schrittweise? Soll die Ladespannung langsam auf den neuen Wert gefahren werden, oder verbirgt sich noch mehr dahinter?
Wäre es nicht besser dies als eine allgemeine Funktion zu machen und nicht an ein 'Charge Current Cut Off' zu binden?

Svol0, 2025-03-15:
Ich möchte mich hier gerne einklinken und hoffe, dass @mmoe-milano nichts dagegen hat.
Die Funktion der sanften Spannungsänderung stelle ich mir wie folgt vor:
- Jede Änderung der Zielspannung ungleich der „Float Ladespannung“ soll durch „sanftes Rampen“ auf die Zielspannung erfolgen. Also in beide Richtungen. Float -> Absorption -> Ballance -> Float.
- Die „Float Ladespannung“ als unteres Minimum muss ohne sanftes Rampen von unten her angefahren werden. (z.B. nach einem Neustart des BSC darf nicht von Null aus langsam auf die Zielspannung gefahren werden, sonder die Float Ladespannung muss direkt gesetzt werden.
- Als „Rampe“ würde ich mit 10mV pro Sekunde anfangen zu testen. Falls möglich wäre ein Parameter für die ersten Versuch hilfreich, mit dem man das Zeitintervall einstellen kann, bis man einen akzeptablen Faktor gefunden hat.

Links zu diesem Thema im Discord Channel:

https://discord.com/channels/1037083515755434005/1342092584201355275/1342966614135410850

https://discord.com/channels/1037083515755434005/1342092584201355275/1346253951267508255

https://discord.com/channels/1037083515755434005/1342092584201355275/1347326237634199584

Fuendiger, 2025-03-17:
10 mV und 1 s sind auch die kleinste Einheiten die möglich sind.
Ich habe es einmal in den Backlog verschoben ;-)

mmoe-milano, 2025-03-18:
@Fuendiger @Svol0 Hallo tut mir Leid! Eure Beiträge zum eröffneten issue sind mir entgangen sonst hätte mich schon früher gemeldet. Zum Verständnis, ich hatte schon mehrere interessante Beiträge zum Thema DC Netzrückspeisung im BSC Forum gelesen. Die Problematik die meines Erachtens dabei entstehen kann ist dass bei aktivierten DC-Feedin und plötzlicher Reduktion der Steuerspannung durch die CutOff Funktion bei den Victron MPs ein gewaltiger Ausgleichstrom fliesen kann, da die Akkus ja noch eine höhere Ladespannung besitzen. Um nicht auf DC-Feedin gänzlich verzichten zu müssen oder mit einem ausgelagerten Workaround in NodeRed leben zu müssen war die Idee dies über den BSC zu erledigen und zwar so dass man den 'setpoint-cv' Parameter (ausgelöst durch CutOff) über die Zeit 'langsam' reduziert. Dadurch würde der Entladestrom aus den Akkus nicht zu groß werden. Dies wäre durch die Parametrisierung eines Rampenverhaltens zur Senkung des Spannungswertes von 'setpoint-cv' über die Zeit erfüllbar.

tueftler0, 2025-04-16:
Wenn die Funktion implementiert ist, sollte man die Problematik aus https://discord.com/channels/1037083515755434005/1336305383815512186 damit einmal testen.

closed 2026-01-17 by shining-man