shining-man/bsc_fw #240: Status Display flackert, seit Update auf FW.0.8.0_T9

Emilleopold, 2025-04-20:
Hallo,

grundsätzlich läuft mein BSC mit der FW 0.8.0_T9. Allerdings habe ich jetzt festgestellt, dass das Status Display flackert. D.h. es geht alle paar Sekunden aus und zeigt dann kurz wieder was an. Bei der FW 0.7.x hat es noch funktioniert. Kann das mit der FW 0.8.0 zu tun haben ?
Auf dem Status Display ist die FW 1.0.0 drauf. Muss ich hier auf FW 1.0.3 updaten, um das zu beheben ?
Danke für eine entsprechende Info.

tueftler0, 2025-04-20:
Hallo.
Das liegt an der inkompatiblen Firmware zu den Sponsor-Versionen.
Im Chat gibt es vom 26.02.2025 unter Teststände eine kompatible zu installieren.
Diese kannst Du für das Display verwenden:
https://discord.com/channels/1037083515755434005/1325511944047824948/1344397456707682369

Fuendiger, 2025-04-20:
Da war ich wohl zu langsam ;-)
War euch auch gerade am Schreiben.

tueftler0, 2025-04-20:
@Emilleopold  Du kannst die Version nun auch im [Display-Release](https://github.com/shining-man/bsc_display/releases/tag/v1.0.4) herunter laden.

Emilleopold, 2025-04-20:
Habe das Status Display upgedated und es funktioniert, Danke.

closed 2025-04-20 by Emilleopold