shining-man/bsc_fw #246: Alterung von Zellen erkennen über einknicken der Ladeendspannung

tueftler0, 2025-07-06:
Evtl wäre es in einer späteren Version interessant das Thema zu behandeln.

Weitere Infos im Discord:
https://discord.com/channels/1037083515755434005/1384941102058045582

shining-man, 2025-07-07:
Ich habe einmal den Beitrag aus dem Discord hier reinkopiert, damit man weis, worum es geht:

> Ein sehr interessanter Beitrag, bei dem es um eine Überladung von alternden Zellen geht. Ich **denke** die Info passt auch analog zu unseren LiFePo4-Typen.
> 
> Evtl wäre es interessant, dies in den BSC zu integrieren - Was meint Ihr darüber?
> 
> Bei den 18650ern sinkt die Ladeendspannung über das Alter.
> Es gibt in der Ladespannung einen negativen Knick, obwohl weiterhin der Ladestrom an liegt. Genau ab dem Punkt verheizt die Zelle intern die Leistung. Diese Stelle bewegt sich immer weiter runter in der Spannung, je nach Alterung.
> 
> -> https://youtu.be/rLIDRAmD6ek?si=9r_0_rasp-gbsgmO
> 
> [image]

Bevor man so etwas umsetzt muss man natürlich wissen ob es sich bei LFP Zellen auch so verhält und in welche Maße. Hier wird es viel Aufzeichnungen brauchen die man analysieren muss, bevor man so etwas umsetzen kann.
Für NCA könnte man dies machen, aber wieviele verwenden diese? verhält sich NMC genauso? NMC und NCA haben verwandte Strukturen, aber verhalten sie sich genauso?

Ein Temperaturensor zwischen jede Zelle und so etwas wird sicher erkannt. Egal welcher Batterietyp.

shining-man, 2025-07-27:
Wenn man den folgenden Screenshot anschaut, dann sieht man warum es nicht so einfach ist, daas zu erkennen:
[image]

Die Zellspannung geht zurück, trotz positivem Ladestrom. Es muss in der Betrachtung also auch berücksichtigt werden, ob der Ladestrom rückläufig ist. Ich denke, dass dies ein Fall für ML ist. Hierzu bräuchte ich aber viele Aufzeichnungen von Ladeverläufen um ein brauchbares Modell zu trainieren. Aber interessant wäre es.

tueftler0, 2025-07-27:
An einem Balancing-Tag müsste das anders betrachtet werden, ja.

Normal sieht das bei mir so aus:

[image]

<br><br>
Balancing-Tag:

[image]

shining-man, 2025-07-27:
Von normalen Tagen und Balancing Tagen. Und das von unterschiedlichen Leuten. Denn jeder kann mit anderen Spannungen arbeiten. Was bei dir ein Balacing Tag ist, ist bei anderen vielleicht ein normaler Tag. 
Super wäre es dann noch neben den Guten auch ein paar Aufzeichnungen von defekten Zellen zu haben.