shining-man/bsc_fw #247: MQTT Sendefrequenz

holle75, 2025-07-25:
Eine Verständnisfrage: Gibt es einen technischen Hintergrund, warum man als User die Sendefrequenz MQTT nicht unter 30 Sekunden stellen kann?

Bei meinem aktuell genutzten Projekt (syssi) kann ich da auch bis 5 Sekunden (oder mehr?) runter und war schon am Überlegen, ob mich die Cell voltages nicht vielleicht doch sogar alle 2 Sekunden interessieren würden. Von der Last her ist das kein Problem.

Warum die Einschränkung?

shining-man, 2025-07-26:
Das versenden von MQTT Daten ist nicht die Hautpaufgabe des BSC. Alle wichtigen Nachrichten wie z.B. Strom, Spannung der Batterie wird ca. jede Sekunde gesendet.

Noch dazu wird es sportlich jede Sekunde 1000 Nachrichten über MQTT zu versenden, wenn man mehr als nur dies machen will. Der BSC hat hier eine Limitierung und darf nur alle 15ms eine Nachricht senden.

holle75, 2025-07-26:
Verstehe. Ich logge zB die Zellspannungen. Werden diese auch sekündlich gesendet? Bin noch in der Trockenphase, Vorbereitungen, um all die anderen Verbindungen die es ansonsten parallel geben müßte mitzuplanen.... oder zu wissen, was ich an Bestehendem eliminieren kann.

Mir würden schon die paar Infos sekündlich (oder wählbar als separate Einstellung) reichen, die unter "Live Daten" zu sehen sind. 

[image]

Falls diese nicht sekündlich/einstellbar kommen, braucht es noch ein zusätzliches "Gerät", in meinem Fall das syssi Projekt. Mehr Hardware, mehr konfigurieren, mehr von allem ... ohne wirklichen Grund. Denn auch wenn der BSC dafür nicht "gemacht ist", warum eine Funktionalität beschneiden? .... und die paar Werte werden den Server nicht wirklich überlasten.

holle75, 2025-07-26:
warum nicht zusätzlich

- Zellspannungen An/Aus
- Zellspannungen Intervall (hier kann man ja wirklich ein unteres Limit von 1 Sek definieren)
- Zellspannungen Delta für Message (zB 20mV. So reduziert man die Datenmenge deutlich)

die unteren außer "bmsStateFETs", "BMS errors" braucht man häufiger als alle 30 Sekunden (ich weiß noch nicht welche sowieso schon öfter kommen)

shining-man, 2025-07-26:
Es überlastet natürlich nicht der Broker, aber den BSC beschäftigt es dann um so mehr. Wie oben geschrieben geht es bei alle Werten um ca. 1000 Nachrichten. Auch nur die Zellspannungen wären schon über 400 Nachrichten. Und das dann jede Sekunde. Nee, der BSC hat eine andere Aufgabe. Wenn du jede Sekunde diese ganzen Werte brauchst, dann wirst du dir stand heute eine andere Lösung suchen müssen. Ob sich in Zukunft da noch was ändert lasse ich offen, aber ich glaube es nicht. Wo ist der Sinn, die Werte jede Sekunde zu haben?
Du kannst höchstens versuchen ob du sie jede Sekunde über die Restapi holen kannst.

holle75, 2025-07-26:
Der Sinn insbesondere die Zellwerte häufiger als alle 30 Sekunden zu bekommen liegt im Logging. Der eine macht/braucht das nicht viele wollen/machen das. Die Foren sind voll mit grafischen Zellspannungswerten. Damit das Sinn macht brauchst du eine niedrige Frequenz. Und ja, die Aufgabe des BSC ist eine andere. Aber warum nicht die eierlegende Wollmilchsau basteln, wenn es keinen Grund dagegen gibt. Ist der BSC mit Standardaufgaben schon so am Limit? warum nicht dem User ein zusätzliches Device ersparen.

Restapi wenn du MQTT sowieso schon am Laufen hast wäre ungalant ;)

Ich vermute mal (ganz arg ins Blaue rein), dass 50% der User den BSC in Kombi mit einer Hausautomation einsetzen. Da sind Daten eine goße Freude. Oder das Datensammeln mit einem anderen Device schon exzessiv betrieben haben und der BSC dazugekommen ist (wie bei mir). Sei es für eine grafische Verarbeitung oder, oder .... lass doch den User entscheiden (solange BSC nicht am Limit, dass weiss ich nicht) was er braucht oder nicht. Ich schaue mir zB meinen Zelldrift tatsächlich sehr gerne an über Zeit um manuell darauf reagieren zu können (Schraube Sensorkabel nachziehen, Zelle tauschen, ecc, ecc) ;)

tueftler0, 2025-07-26:
@holle75 
Ich denke auch, dass mindestens 50% der User eine HA zusammen mit dem BSC nutzen.
Total Voltage, Total Current kommen sekündlich.
Warum bräuchtest du Zellspannungen schneller wie es nun integriert ist?
Man kann mit den Daten, wie es nun implementiert ist, die Traces sehr gut darstellen, wie Du sie überall in den Foren gesehen hast.
Ich würde sagen, probiere es einfach mal live aus.
Ein Anschauen des Zelldrifts mache ich auch - Aber auch hierfür reichen die Infos doch wunderbar in der Geschwindigkeit wie es nun ist.

Restapi ist im BSC ohnehin auch integriert: https://bsc-org.github.io/bsc/restapi/

holle75, 2025-07-26:
Hallo! Cellvoltages kommen alle 30 Sekunden. Da mich (und wahrscheinlich andere) vor allem die Spitzen (nach oben und unten) interessieren ist das eine recht nutzlose Frequenz. Wie geschrieben ist die API wenn du schon via MQTT implementiert hast nicht wirklich eine sinnvolle Option. Dann lass ich lieber den syssi-ESP weiter mitlaufen (und genau darum geht es. Warum zwei Kästchen die du administrieren, updaten, etc. musst, wenns auch eins könnte).

Aber das ist ja nur ein Vorschlag gewesen.... den ich zumindest völlig logisch finde. Der Rest war bis jetzt (läuft seit paar Stunden) ein Träumchen. Danke dafür an euch. So einen schmerzfreien Switch von BMS -> Anlage zu BMS -> BSC -> Anlage hatte ich nicht erwartet. Das hat einfach funktioniert. Mal schauen was beim nächsten "Batterie voll"-Moment passiert.

tueftler0, 2025-07-26:
Hi.
Freut mich, wenn es bei Dir so direkt funktioniert hatte!
Dann wünsche ich Dir einmal viel Spaß damit :)

Um den Traffic weiterhin einzugrenzen und trotzdem bei Änderungen aktuell zu bleiben, könnte man natürlich auch statt einem fixen Intervall, ein Event gesteuertes Senden integrieren. Soll heißen: Wenn Werte innerhalb einer Zeit um eine bestimmte Höhe sich ändern, wird gesendet.

tueftler0, 2025-07-26:
Stimmt, habe ich überlesen.

shining-man, 2025-09-29:
Ich habe im aktuellen Teststand T0.9.0_T1 Änderungen vorgenommen.
Theoretisch können jetzt Zeiten ab 1 Sekunde eingestellt werden. Aber es gibt keine Garantie, dass er Zeiten unter 30 Sekunden halten kann. Er sendet dann so schnell er kann.

holle75, 2025-09-30:
Top. Danke. Sobald ich das testen kann, werde ich berichten ...

closed 2025-10-05 by shining-man