shining-man/bsc_fw #251: Unverständliche Triggerreaktion Undervoltage

holle75, 2025-07-26:
Wenn ich den Trigger 5 so definiere:

[image]

und ihn hier als Aktor nutze 

[image]

schaltet es mir direkt den Inverter aus mit Undervoltage Warning. Dies auch wenn die Spannungsschwelle nicht erreicht ist. Jemand eine Idee dazu?

holle75, 2025-07-26:
Der Trigger 5 ist glaubs auch laut Dashboard aktiv. Nur warum?

[image]

shining-man, 2025-07-27:
Weil deine Zellspannung-Max mit sicherheit überschritten ist ;-)
Wenn die nicht greifen soll, dann musst du sie schon auf einen hohen Wert wie z.B. 5000 mV stellen.

holle75, 2025-07-27:
Mmh, war da nicht was mit "0" gleich "gilt nicht"? ;)

closed 2025-07-27 by holle75