shining-man/bsc_fw #255: Feature Request: Zellprofilbasierte Temperatur-/Stromregelung im BSC

D3R-ST3FAN, 2025-09-18:
Kurzbeschreibung:  

Statt einer rein linearen Stromdrosselung ab fixer Temperatur soll der zulässige Lade-/Entladestrom aus temperaturabhängigen C-Raten abgeleitet werden (Zellhersteller-/Zelltypspezifisch; z. B. EVE/Envision). 
Ziel: präzisere, herstellerkonforme Regelung – insbesondere bei niedrigen Temperaturen z.B. Garage/WoMo.

Thread: Discord https://discord.com/channels/1037083515755434005/1410238040693014669  
 
Hinweise/Erkenntnisse:
  - shiningman: MVP könnte als 2-Punkt-Regelung je Richtung (Charge/Discharge) umgesetzt werden (Start/End-C-Rate → linear dazwischen).  
  - Tueftler: Alternative tabellenbasierte Eingabe analog Datenblatt (z. B. bis zu 10 Temperaturfenster je Richtung, je `t_min/t_max` + `C-Rate`; „letzte Definition gilt fort“).

Grafisch wäre vielleicht overkill
---

## Problem
Die aktuelle Logik regelt ab einem Startpunkt linear auf 0 A herunter, unabhängig von Zellchemie/Herstellerdaten. Datenblätter definieren jedoch Temperatur→C-Rate-Profile, die teils deutlich von einer einfachen Gerade abweichen (z. B. EVE 280 Ah).

## Ziel (MVP)
- Zwei-Punkt-C-Rate je Richtung (Charge/Discharge):  
  Konfiguration z. B. `T_min / C_at_Tmin` und `T_max / C_at_Tmax`; dazwischen linear interpolieren.
- Automatische Skalierung in Ampere: `I_max = C_rate × Packkapazität (Ah)` → funktioniert auch, wenn Packs offline sind.
- Temperaturquelle wählbar (bestehende Sensorik; Aggregation Min/Avg/Max).
- Fallback: Wenn nicht konfiguriert → heutiges Verhalten.
- Minimal-Logging: Wechsel des wirksamen Temperatursegments + resultierender Strom.

## Vielleicht später
- SoC-Abhängigkeit integrieren 
- Tabellenbasierte Profile mit mehreren Temp-Fenstern je Richtung (z. B. 5–10 Punkte; Felder `t_min`, `t_max`, `c_rate`; offene Ränder erlaubt; letzte Definition gilt fort) als Alternative zum  grafischen Editor/„Kurven-Drawer“.
- SoC-abhängige Limits (z. B. strenger bei sehr niedrigem SoC).
- Import/Export (CSV/JSON) von Profilen und UI-Vorschau der resultierenden C-Rate-Kurve.

shining-man, 2025-09-20:
So in etwa wird es dann ausschauen. Zusätzlich können die Werte tabellarisch eingeblendet werden. 

[image]

@D3R-ST3FAN Warum einen Fallback? Es reicht doch eine Regelung. Im Prinzip ist das nicht anders als vorher, nur dass es jetzt nach der C-Rate geregelt wird.

Die Seonsoren der Data-Devices kann man ja heute schon auswählen. Als Data-Devices werden die konfigurierten genommen, die auch für alle anderen Regelungen genommen werden.

dfarin, 2025-09-20:
@shining-man Naja, für mich würde die aktuelle Regelung völlig reichen, jedoch mit der zusätzlichen Auswahl der OW Sensoren.
Laden Entladen soll gedrosselt, bzw gestoppt werden zwischen 35-45 Grad. Die C-Rate ist für den Sicherungsschalter natürlich nicht relevant, 
Ich hatte das issue nun so verstanden das es die Möglichkeit der Zell Profile zusätzlich gibt, die alte Logik jedoch nicht rausfällt, diese wäre hier für mich natürlich geeigneter. Die C-Rate Logik wäre hier sogar kontraproduktiv da ich damit deutlich über Limit wäre,

Mir war/wäre nur wichtig das die OW Sensoren mit als Quelle für die aktuelle Logik verfügbar werden.
Tatsächlich würde es damit sogar noch deutlich mehr Möglichkeiten zur Überwachung außerhalb des Akkupacks geben.
Wo ich so n OW Sensor hinhänge ist ja mehr oder weniger völlig frei wählbar. Kühlrippen WR, Lynx, an ein Kabel meiner Wahl, Zimmerdecke, AC Unterverteilung usw.

D3R-ST3FAN, 2025-09-21:
> [@D3R-ST3FAN](https://github.com/D3R-ST3FAN) Zum Punkt `Aggregation Min/Avg/Max`: Aktuell ist es so, dass er immer den Min/Max Wert nimmt, je nach dem in welche Richtung. Von warm nach kalt nimmt er den Min-Wert und von kalt nach warm den Max-Wert. die Regelung gerade geht. Wenn man dies einstellbar machen möchte, mit welchen Optionen? Ich möchte das ganze jetzt nicht mit zuviel Optionen unnötig kompliziert machen. Man könnte zwei Optionen machen: Min/Max und Avg.

Perfekt, das hatte ich sag gar nicht auf dem Schirm. Sollte ja passen.
Ist glaube ich genau, was ich meinte.

Wenns Richtung Kalt geht wird die niedrigste Temperatur genommen (Min Wert) und in Richtung warm die höchste Temperatur (Max Wert).

D3R-ST3FAN, 2025-09-21:
> So in etwa wird es dann ausschauen. Zusätzlich können die Werte tabellarisch eingeblendet werden.
> 
> <img alt="Image" width="707" height="354" src="https://private-user-images.githubusercontent.com/109786660/491965834-a661f612-8c4c-4e85-9538-d1cdca7fc6b5.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg0NzY5OTIsIm5iZiI6MTc1ODQ3NjY5MiwicGF0aCI6Ii8xMDk3ODY2NjAvNDkxOTY1ODM0LWE2NjFmNjEyLThjNGMtNGU4NS05NTM4LWQxY2RjYTdmYzZiNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkyMVQxNzQ0NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00YTRkNDFmNjJiN2ZlNDhlN2I0NTMzNzFlMTE0NWVjYWVjNjZkMDVlMjU3YWZkOWYzZDYwOWU1NGY1OTg3MzQwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.xbt7fE_sWCXfKdXH4BvfyfOPIPOyBmIjhoerFbEvXh4">
> [@D3R-ST3FAN](https://github.com/D3R-ST3FAN) Warum einen Fallback? Es reicht doch eine Regelung. Im Prinzip ist das nicht anders als vorher, nur dass es jetzt nach der C-Rate geregelt wird.
> 
> Die Seonsoren der Data-Devices kann man ja heute schon auswählen. Als Data-Devices werden die konfigurierten genommen, die auch für alle anderen Regelungen genommen werden.

Fallback war so gemeint, dass wenn man den "Zellprofil" Modus nicht verwendet, der "alte" aktiv ist.

shining-man, 2025-09-29:
Im Teststand T0.9.0_T1 ist eine erste Version umgesetzt.
Was noch fehlt, ist die Auswahl der Onewire Sensoren und die Wahl zwischen Min/Max und Avg..

D3R-ST3FAN, 2025-10-23:
Danke für den Hinweis mit der Rest-API die nutze ich kaum.
Ich wäre aber auch sicher nicht auf "cc_temperatureP" gekommen.

Was ich aber eigentlich meinte/suche ist der errechnete Wert.
"cc_temperatureP" ist scheinbar immer MAX solage der Wert aus der Regelung über Max ist.

D3R-ST3FAN, 2025-11-18:
Die Balkenanzeige funktioniert bei der T0.9.4 Insider.

[image]

[image]

Was ich mich gerade Frage. Bei der Einstellung "Charging current per pack too high
Throttle charging current if the charging current of a pack is exceeded".
Wenn das aktiviert wird werden dann die Werte aus temperaturabhängigen Steuereung (C-Profil) ebenfalls genutzt?

Also quasi die Frage ob das pro das ebenfalls pro Batterie berechnet wird oder nur als Summe.

Falls das (noch) nicht der Fall ist, sollte es vielleicht ins Backlog.
Oder wäre das etwas für ein weitres Issue?

shining-man, 2025-12-26:
Ich habe mich jetzt doch für die höchste C-Rate entschieden. Gibt es mehrere Punkte mit gleich hoher C-Rate wird der erste genommen.

D3R-ST3FAN, 2026-02-03:
Für mich funktioniert das jetzt wie erwartet. Zumindest in Richtung niedrige Temperatur konnte ich positv testen.
Vielen Dank!